duinoWitchery/hd44780 #2: Hello World example - variable overflow

Bodmer, 2016-12-09:
Variable "mins" is an unsigned int and will overflow before millis() does.

Formatting on LCD only allows hours to be 99 max before character shift right and leaving unwanted digits after overflow. Overflows will also cause "millenium bug" style time discontinuities in displayed elapsed time.

OK, it takes a long time to overflow, but Hello World should set a good example and be bug free, it makes one question the rigor that might have been applied within the library :-)

bperrybap, 2016-12-09:
I agree. - To b honest it has kind of bothered me for a while as well.
I'll fix it to work correctly, at least until millis() rolls over - which all the other helloworld examples will see as well.

It is a pretty simple change to fix the mins issue and for 3 digit hours.
I'll toss in a few more comments into the lower level code functions.

bperrybap, 2016-12-27:
Now I remember why I limited this to only 2 digits for hours.
This was to allow it to work on 8x2 displays.
Still thinking about the final solutions but may mod the hours with 100 to limit it to 0-99

closed 2017-10-20 by Bodmer