duinoWitchery/hd44780 #6: Line Wrap Demo omits first character after lcd.clear()

cerkit, 2018-08-19:
In the Line Wrap example for the I2Cext section, I am noticing that when `lcd.clear()` is called, the next character printed is omitted. I have a workaround of just printing a space, but I thought I would report it.

Here is my sample code:
``` c
lcd.clear();
lcd.print("WrapTest");
```

This displays `rapTest` on the LCD display.

Also, the `lcd.cursor()` method is ignored after `lcd.clear()` is called. I was able to make it work by calling `lcd.cursor()` twice.

I have a 20x4 LCD display with the I2C backpack.

bperrybap, 2018-08-19:
I am not seeing this behavior when testing with several different LCD displays and different backpacks including both PCF8574 and MCP23008 i/o expanders.

It is possible that you have a LCD that is slower than the reference timing that needs additional time to complete commands.

In order for me to replicate your test, I need the full sketch you are using.

Also, Can you run the I2CexpDiag sketch to make sure that there are no issues and post the output here.

bperrybap, 2018-08-19:
Actually, can you start a thread on the Arduino forum in the Displays section so we can further discuss.
The eventual solution may also help others.

bperrybap, 2018-08-19:
I'd like to leave this open until it is resolved.
I just wanted to move the conversation to the Forum while we narrowed it down.

bperrybap, 2018-08-20:
Issue was due to incorrect Arduino board type selected in IDE with 8Mhz clock instead of a board type with a 16Mhz clock that the board was actually using.
Arduino thread about it is here: https://forum.arduino.cc/index.php?topic=564426.0

closed 2018-08-20 by bperrybap