duinoWitchery/hd44780 #8: Problems when creating hd44780_I2Cexp{} instances dynamically

PatrykSSS, 2019-05-28:
Display data gets corrupted (lines begin in some random positions on the display) when instances are created with operator 'new'. No problems when created statically (as global variables). I'm using two 20x4 LCDs.

bperrybap, 2019-05-28:
Can you provide a minimal working example so that I can see and replicate the problem.

PatrykSSS, 2019-05-29:
`#include <Wire.h>
#include <hd44780.h>                       // main hd44780 header
#include <hd44780ioClass/hd44780_I2Cexp.h> // i2c expander i/o class header

hd44780_I2Cexp * pLcd0 = NULL;
hd44780_I2Cexp * pLcd1 = NULL;
hd44780_I2Cexp * pLcd2 = NULL;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(2000000); // (115200);
  Serial.println(F("\r\nHello"));
  Serial.println(F("Built: " __DATE__ " " __TIME__ ".\r\n"));
  pLcd0 = new hd44780_I2Cexp(0x27, I2Cexp_PCF8574, 0,1,2,4,5,6,7,3,HIGH);
  Serial.print(F("Using I2C LCD_2004 (through PCF8574A on 0x27), init status: ")); Serial.print(pLcd0->begin(20, 4)); Serial.println(F("."));
  pLcd1 = new hd44780_I2Cexp(0x26, I2Cexp_PCF8574, 0,1,2,4,5,6,7,3,HIGH);
  Serial.print(F("Using I2C LCD_2004 (through PCF8574A on 0x26), init status: ")); Serial.print(pLcd1->begin(20, 4)); Serial.println(F("."));
  pLcd2 = new hd44780_I2Cexp(0x3E, I2Cexp_PCF8574, 0,1,2,4,5,6,7,3,HIGH);
  Serial.print(F("Using I2C LCD_1602 (through PCF8574A on 0x3E), init status: ")); Serial.print(pLcd2->begin(16, 2)); Serial.println(F("."));
  // Print a message to the LCD.
  pLcd0->print(F("Hello Hello Hello He")); pLcd0->setCursor(0, 1); pLcd0->print(F("12345678")); pLcd0->setCursor(0, 2); pLcd0->print(F(__DATE__ " " __TIME__));
  pLcd1->print(F("Hello Hello Hello He")); pLcd1->setCursor(0, 1); pLcd1->print(F("12345678")); pLcd1->setCursor(0, 2); pLcd1->print(F(__DATE__ " " __TIME__));
  pLcd2->print(F("Hello Hello Hell"    )); pLcd2->setCursor(0, 1); pLcd2->print(F("12345678"));
}

void loop() {
  // put your main code here, to run repeatedly:

}`
[image]

PatrykSSS, 2019-05-29:
Here no repro:
`#include <Wire.h>
#include <hd44780.h>                       // main hd44780 header
#include <hd44780ioClass/hd44780_I2Cexp.h> // i2c expander i/o class header

hd44780_I2Cexp Lcd0(0x27, I2Cexp_PCF8574, 0,1,2,4,5,6,7,3,HIGH); // declare lcd object - hardcoding values detected by I2CexpDiag
hd44780_I2Cexp Lcd1(0x26, I2Cexp_PCF8574, 0,1,2,4,5,6,7,3,HIGH); // declare lcd object - hardcoding values detected by I2CexpDiag
hd44780_I2Cexp Lcd2(0x3E, I2Cexp_PCF8574, 0,1,2,4,5,6,7,3,HIGH); // declare lcd object - hardcoding values detected by I2CexpDiag

void setup() {
  // put your setup code here, to run once:
  Serial.begin(2000000); // (115200);
  Serial.println(F("\r\nHello"));
  Serial.println(F("Built: " __DATE__ " " __TIME__ ".\r\n"));
  Serial.print(F("Using I2C LCD_2004 (through PCF8574A), init status: ")); Serial.print(Lcd0.begin(20, 4)); Serial.println(F("."));
  Serial.print(F("Using I2C LCD_2004 (through PCF8574A), init status: ")); Serial.print(Lcd1.begin(20, 4)); Serial.println(F("."));
  Serial.print(F("Using I2C LCD_1602 (through PCF8574A), init status: ")); Serial.print(Lcd2.begin(16, 2)); Serial.println(F("."));
  // Print a message to the LCD.
  Lcd0.print(F("Hello Hello Hello He")); Lcd0.setCursor(0, 1); Lcd0.print(F("12345678")); Lcd0.setCursor(0, 2); Lcd0.print(F(__DATE__ " " __TIME__));
  Lcd1.print(F("Hello Hello Hello He")); Lcd1.setCursor(0, 1); Lcd1.print(F("12345678")); Lcd1.setCursor(0, 2); Lcd1.print(F(__DATE__ " " __TIME__));
  Lcd2.print(F("Hello Hello Hell"    )); Lcd2.setCursor(0, 1); Lcd2.print(F("12345678"));
}

void loop() {
  // put your main code here, to run repeatedly:

}`
[image]

bperrybap, 2019-05-29:
What Arduino core and board type are you using?

bperrybap, 2019-05-29:
Was able to reproduce the issue.
The issue is the _wraplines variable flag was not being explicitly cleared. On static it gets cleared automatically but new does not clear the object data.

Before I do create a release , could you try this to make sure it fixes the issue you having.
Modify the constructors in hd4470.cpp to clear _wraplines like this:
```
hd44780::hd44780()
{
	/*
 	 * Set up default execution times for clear/home and
	 * all other instructions and data writes.
	 * Setting exec times here
	 * allows application to call setExecTimes() before begin()
	 * should it be necessary to override defaults.
	 */

	setExecTimes(HD44780_CHEXECTIME, HD44780_INSEXECTIME);
	_wraplines = 0;

	markStart(0); // initialize last start time to 'now'
}

hd44780::hd44780(uint8_t cols, uint8_t rows) : _cols(cols), _rows(rows)
{
	setExecTimes(HD44780_CHEXECTIME, HD44780_INSEXECTIME);
	_wraplines = 0;
	markStart(0); // initialize last start time to 'now'
}

hd44780::hd44780(uint8_t cols, uint8_t rows, uint32_t chExecTimeUs, uint32_t insExecTimeus) :
		 _cols(cols), _rows(rows), _chExecTime(chExecTimeUs), _insExecTime(insExecTimeus)
{
	_wraplines = 0;
	markStart(0); // initialize last start time to 'now'
}
```
Let me know if this fixes the issue.

PatrykSSS, 2019-05-29:
Arduino Nano with ATmega328P.

PatrykSSS, 2019-05-29:
Didn't seem to help. I just noticed that the issue is not 100% reproducible - now I needed to cycle power to Arduino few times to get repro, and when I did the issue is 100% reproducible across non-power cycle resets (also across reflashing FW).

PatrykSSS, 2019-05-29:
Suspicion about uninitialized data is valid - putting instances on the stack also reproduces the problem.

bperrybap, 2019-05-29:
I was doing some testing right as you posted.
The problem in the hd44780 code is definitely due to it assuming the object was zeroed out and new does not initialize the object to zero. There are few places in the hd44780 code that assume it was zeroed out. But the fix is more complicated than just clearing that single _wraplines variable.
There are other things that I need to fix & resolve for some more complex usage cases in other i/o classes.
For the time being you should be able to use new in your code use case by simply updating the first constructor in hd44780.cpp like this to zero out the object.
```
hd44780::hd44780()
{
	memset(this, 0, sizeof(*this));
	/*
 	 * Set up default execution times for clear/home and
	 * all other instructions and data writes.
	 * Setting exec times here
	 * allows application to call setExecTimes() before begin()
	 * should it be necessary to override defaults.
	 */

	setExecTimes(HD44780_CHEXECTIME, HD44780_INSEXECTIME);
	markStart(0); // initialize last start time to 'now'
}
```
and I'll go off and work on the real fix.

bperrybap, 2019-05-29:
Out of curiosity, what type of application are you doing that is using/needing dynamic LCD objects?

PatrykSSS, 2019-05-29:
It works indeed. Wouldn't it harm base class - Print? Its constructor is called before hd44780::hd44780(), I didn't look in Print implementation though.

PatrykSSS, 2019-05-29:
As of dynamic objects - it's just the only way I found so far to switch I2C address of the LCD without using autodetection routines. I'll have just one LCD, but depending on I2C chip (PCF8574 or PCF8574A, out of control when buying cheap LCDs) the address is different.

bperrybap, 2019-05-29:
Yeah, not sure how the memset() might affect the base class. I don't plan on using it in the final solution.
Why can't you use the auto detection & auto configuration? The issue you talk about is exactly what it is designed to resolve. i.e. plug in the LCD, and the library figures out what you have and uses it.
It not only optionally detects the i2c address but also the pin mappings and backlight active level which can also vary from backpack to backpack. 
Since the initialization code runtime looks at the constructor parameters, it won't save any code space to use the full/manual constructors. i.e. all the auto detect and auto configure code will get linked in anyway.

bperrybap, 2019-05-29:
When I said "auto detection" etc... I meant the capability built into the hd44780 library.
In other words, why can you let the library do all the auto detection and auto configuration of the LCD backpack for you?

PatrykSSS, 2019-05-30:
Maybe clear with memset() just form the offset equal to base class size?
I wanted to save code space and "boot" time while having the same lib which I used to autodetect parameters once (autodetect every I2C converter kind once - on my desk), but it seems I'll save just the time, I'll think about it.

bperrybap, 2019-05-30:
I don't think memset() is an option. The language really doesn't guarantee the memory layout of the object and it doesn't work for some of the other use cases anyway.  There are other ways to handle it that I'll use.

bperrybap, 2019-05-30:
The code has been updated to support using "new" for lcd objects.
The code is on the alphadevel branch. It is an update to hd44780.cpp
Here is the code: https://github.com/duinoWitchery/hd44780/blob/alphadevel/hd44780.cpp
Give it a try. It should resolve all your issues.

PatrykSSS, 2019-06-13:
I confirm it works - tried 10 times on the same HW which failed today (100%) with previous version. Sorry that so late - quite busy time recently. Thanks! :-)

bperrybap, 2019-06-13:
FYI, One thing you should know, is that the current code update created an issue. The cursor addresses/locations will be wrong for anything but a a 16x2 display. i.e. setCursor() will position to the wrong location. I've fixed this but have not pushed up the code yet as I'm trying to shrink the code a little bit as I'm not happy yet with the added code size to resolve things for using new on the object.

I'm still curious why you can't use a static object with auto auto address location?
If you use an empty constructor it does everything for you. If you use an address of zero it will auto locate the i2c address.

PatrykSSS, 2019-06-13:
>> The cursor addresses/locations will be wrong for anything but a a 16x2 display. i.e. setCursor() will position to the wrong location.
I didn't noticed it, I'll check again.

>> I'm still curious why you can't use a static object with auto auto address location?
I can, just wanted to save some time for autodetection, I'll compare how much and maybe will use it.

bperrybap, 2019-08-12:
fixed in latest release (1.1.0) 
Code no longer assumes certain variables are automatically zeroed.
There were two commits for this, but the latest commit 4fd309b129749e3e2e1cf6c65530d802eec8b367 is the final fix for this.

closed 2019-08-12 by bperrybap