duinoWitchery/hd44780 #12: lcd.clear() in linewrap test causes 1st char of 1st line to be missed.

alw1746, 2020-01-01:
See comments in attached linewrap sketch.

 /* if lcd.clear() is enabled, the letter A in first line goes missing and line looks like "23456789 123456789W ". All other lines are fine.
  * This behaviour is present even if linewrap is disabled.
  * if lcd.clear() is disabled, all lines display correctly - "A23456789 123456789W", etc.
*/

[linewrap.txt](https://github.com/duinoWitchery/hd44780/files/4013997/linewrap.txt)

bperrybap, 2020-01-01:
I can't reproduce the issue. It displays ok for me.
Please provide additional details about your h/w & s/w tools.
i.e. the Arduino board you are using,  LCD part number, i2c backpack part number,
IDE version,  and hd44780 library version, 

I would be curious if you add a delay(5) between the clear() and the first print() if the 'A' character shows up.

--- bill

alw1746, 2020-01-01:
The delay(5) did the trick, works fine now. Interestingly the problem does not appear on my 16X2 with the same expander. My guess is the 20x4 is a bit old and slow(bought in 90's). Details:

Arduino Nano V3
Optrex DMC20434 20x4 LCD 14-pin no backlight
generic cheap PCF8574 backpack expander
IDE 1.8.8
hd44780 v1.1.0

Great library and thanks for your help!

bperrybap, 2020-01-01:
I looked at a datasheet for the Optrex DMC20434
It showed the typical execution time for the Clear & Home commands at 15.2ms and all the others at 40us

I started with the more common timing of 1.52ms and 37us from hd44780 datasheet and bumped them a bit to 2ms and 38us for the hd44780 library timing.

The datasheet shows it is considerably slower for the clear and home commands that most other displas.  (seems odd that is is exactly 10x slower than the Hitachi chip timing).

Good news is you can set the timing for each individual display from your sketch.
use the  setExecTimes(chUs, insUs) function to set the timing as desired.

If the datasheet I looked at (page 35): http://www-inst.eecs.berkeley.edu/~cs150/fa01/labs/project/LCD_User_Manual.pdf 
was correct, then use:

```
lcd.setExecTimes(15200, 40);
```

But I'm guessing that it really doesn't take 15.2ms since a simple 5ms added delay seemed to resolve the issue.
So you may want to experiment a bit with the timing a bit if you want it to be faster for additional update performance.

alw1746, 2020-01-01:
lcd.setExecTimes(3000, 40)  or delay(3) worked fine...cheers Alex

bperrybap, 2020-01-02:
If you use delay() it has to be called after every clear() or home() command. 
IMO, it is a kludge/work-around.

The hd44780 library provides a better way.
Use setExecTimes() once in setup() and before begin() to set the timing for that display.

Interesting that 3ms works, when the datasheet shows 15.2ms
Not surprising as I see lots of issues & even errors in LCD datasheets.

alw1746, 2020-01-02:
Agreed. setExecTimes is the way to go.

bperrybap, 2020-01-02:
I'm assuming this issue can be closed as it seems to be an issue of a slow LCD that can be resolved using setExecTimes()

alw1746, 2020-01-02:
No worries I'll close it, thanks again for good advice.

closed 2020-01-02 by alw1746