duinoWitchery/hd44780 #13: Are 20x4 LCDs supported?

jskubick, 2020-03-08:
I bought a 20x4 LCD with PCF8574 backpack (https://amazon.com/gp/product/B01DKETWO2) and tried using the library with hd44780_I2Cexp.h (code follows at end).

It looks like it's not properly handling the 20-column lines. Here's what ultimately gets printed:

Hello, World! This i
eriment with
s a long line to exp
[fourth line is blank]

I remember that 20x4 44780s are a bit funky to deal with when programming directly compared to 16x4, but it's been years since the last time I've had to do it. Does this library support 20x4 LCDs, or do I have to do something different to initialize it besides merely setting LCD_COLS = 20?

Thanks!

(code follows)

#include <Wire.h>
#include <hd44780.h>
#include <hd44780ioClass/hd44780_I2Cexp.h>

hd44780_I2Cexp lcd;
const int LCD_COLS = 20;
const int LCD_ROWS = 4;

void setup() {
  // put your setup code here, to run once:
  int status;
  
  // initialize LCD with number of columns and rows: 
  // hd44780 returns a status from begin() that can be used
  // to determine if initalization failed.
  // the actual status codes are defined in <hd44780.h>
  // See the values RV_XXXX
  //
  // looking at the return status from begin() is optional
  // it is being done here to provide feedback should there be an issue
  //
  status = lcd.begin(LCD_COLS, LCD_ROWS);
  if(status) // non zero status means it was unsuccesful
  {
    status = -status; // convert negative status value to positive number

    // begin() failed so blink error code using the onboard LED if possible
    hd44780::fatalError(status); // does not return
  }

  // Print a message to the LCD
  lcd.print("Hello, World! This is a long line to experiment with");
}

void loop() {
  // put your main code here, to run repeatedly:

}

bperrybap, 2020-03-08:
Yes 20x4 displays are supported.
In terms of handling line wrapping, I would put this in the "ooops, user error" category.
You didn't tell the library to do line wrapping. 
By default the hd44780 library does not do any line processing and hands the characters to the LCD, the same as the LiquidCrystal library. Since the LCD also does not do any line wrapping, it will look odd on the display and likely be unexpected due to how memory is mapped to character positions on the LCD screen.

Take a look at the API documentation, specifically the lineWrap() call and the included LineWrap example to see how to use the lineWrap() function to enable line wrapping.

closed 2020-03-08 by bperrybap

jskubick, 2020-03-08:
(blush) Whoops, thanks!