duinoWitchery/hd44780 #14: Compactible with expander PCA8574?

mohdrais, 2020-04-11:
I tried on expander PCA8574, but it did not work. It just hang. How to make it working?

mohdrais, 2020-04-11:
Sorry forgot to provide detail. Arduino Nano Every was used to run the PCA8574 by using Arduino1.18.2

bperrybap, 2020-04-11:
I've not seen any LCD backpacks that use a PCA8574.
 Is this an actual product or something you have created on your own?
If it is a product, I need to see the details on the backpack product you are using. 
Please provide a link to the product. 
A schematic would also be useful.

If it is something you have created on your own, please provide a schematic and some photos so I can see how everything is wired up.

mohdrais, 2020-04-12:
These are the LCD which I bought from DFROBOT. Both did not work (for Nano
Every and Arduino 1.8.12) but both did not have any problem with UNO. When
I test with I2CexpDiag, LCD 16x2 hangs but LCD 20x4 give the address of
0x20.  I also tested the normal LCD 16x2 by using hd44780_pinIO with Nano
Every, and it works. These products are as follow:
https://www.dfrobot.com/product-590.html (20x4) and
https://www.dfrobot.com/product-135.html (16x2)

bperrybap, 2020-04-12:
Post the output from I2XexpDiag for each so I can see where it is hanging.

mohdrais, 2020-04-12:
Below are the output as requested :
1.I2CLCD2x16.png is the output of I2XexpDiag. I waited for 6 minutes and it
still scanning I2C bus
2.I2CLCD4x20B.png is the output of I2XexpDiag. The result of the scanning
is almost instant.
3.I2CLCD4x20A.png is the output of I2XexpDiag. I purposely connect the SDA
line from the LCD to SCL of Nano Every and viceversa just to test the
I2XexpDiag
4. While running all the above testing, I have to comment the lines as
highlighted in the IDE.

bperrybap, 2020-04-12:
There is no I2CexpDiag output or images in the post.
You can see the github issue thread here: https://github.com/duinoWitchery/hd44780/issues/14

I'd like to understand what is going on and either fix the issue or see if I can work around it.
Lets continue this discussion over on the Arduino Forum as this issue is likely to start to get a bit involved and the information may help others that run into this.

In the Arduino forum thread, for the I2Cexpdiag output, it would be better to post the text output  either inline text or a text file attachment rather than images as it is easier to read.
Start a new thread in the Displays section, and we will continue over there.
https://forum.arduino.cc/index.php?board=7.0
I'll look for the thread over on the Arduino Forum.

mohdrais, 2020-04-13:
Sorry for the problems. Below are the pictures:
1.I2CLCD2x16.png
[image: I2CLCD2x16.png]
2.I2CLCD4x20B.png

[image: I2CLCD4x20B.png]

3.I2CLCD4x20A.png
[image: I2CLCD4x20A.png]
Below are the output tested on UNO. Note: All do not need to be commented
on the INPUT_PULLUP as for Nano Every
1.UNOLCD16x2

[image: UNOLCD16x2.png]
2.UNOLCD20x4
[image: UNOLCD20x4.png]

bperrybap, 2020-04-13:
The images are not attached. But put them in your Arduino thread:
https://forum.arduino.cc/index.php?topic=677156
rather than here so we can continue discussion in that thread.

bperrybap, 2020-06-13:
The PCA8574 appears to be s/w compatible with the PCF8574.
It appears to be just a faster version of the PCF8574.

From information in the Arduino thread, it appears that there is no i2c i/o expander chip issue but rahter it appears that the I2CexpDiag code was modified in a way that broke the code.
The I2Cexpdiag code modification appears to be incorrectly trying to solve a I2CexpDiag compilation issue relate to an issue in the megaAVR core.

As of hd44780 library version 1.2.0, I2CexpDiag no longer checks for the INPUT_PULLUP macro and just assumes that the symbol name exists in the core.

This should fix the compilation issue on the megaAVR core.

bperrybap, 2020-06-13:
Is there still an issue?

mohdrais, 2020-06-13:
Thank you for your concern. I have not tested the new library. Right now I
am concentrating on mechanical design of vibrating platform, maybe after
completing the design then I will inform you. Again, thank you very much.

mohdrais, 2020-06-19:
I have tested your new hd44780 and I am so sorry it still did not work.
Below are the screen capture of I2CexpDiag for 2x16 LCD and 4X20 LCD.
Nothing happened both were tested for ioClass.hd44780_I2Cexp.HellowWorld
[image: LCDI2C4X20.png]
[image: LCDI2C16x2.png]

bperrybap, 2020-06-19:
Saying it doesn't work with no other information and no images is not helpful.
If I can't see an issue, there is nothing for me to resolve or to fix.
I would suggest that you update your Arduino forum thread with any new information you have about any issues you are having.
Describe what "did not work" means.
Include I2CexpDiag output as well as any relevant images that demonstrates the issue.
From looking at the schematic, it appears that the design they used breaks the hd44780 backlight active level detection.
So I'm guessing that it is working ok, but the backlight control is backwards due to their design breaking the auto level detection.
I'll post more about this in the thread once you update it with the information about the issue youare currently having.