duinoWitchery/hd44780 #15: Program design for multiple displays

obdevel, 2020-05-02:
I have a requirement to support up to 32 displays across four I2C buses. I'd be using 1601A, 1602 or 2004 displays with PCF8574T backpacks.

Bus switching is performed by a TCA9546A IC which requires a single I2C write to its own address to select the bus before writing to the I2C display adapter's address. Thus I can have duplicate I2C addresses (0x20-0x27) on each bus. Clearly this means the library's auto locate/autoconfig capability is not possible.

Memory would be an issue if I instantiated a hd44780_I2Cexp object for every display (at 45 bytes per object). Creating an object for each of eight addresses (0x20-0x27) might be acceptable. 

I could recreate a single object each time I want to write to a device (new/delete), but I dislike dynamic memory allocation on small microcontrollers. Execution speed is not critical.

Is there are way to reconfigure a single hd44780_I2Cexp object with a new I2C address ?

Any creative thoughts would be welcome, as would suggestions of a better place to discuss this.

Thank you.

bperrybap, 2020-05-02:
What MicroController are you using?
The AVRs often have pretty limited memory, but most of the other processors have quite a bit more.
A better place to continue the discussion would be the "Displays" section on the Arduino Forum.

obdevel, 2020-05-02:
Thanks Bill. I'll post over there. I'll close this one.

closed 2020-05-02 by obdevel