duinoWitchery/hd44780 #16: Skipping steps while printing

federico77, 2020-05-21:
Hello, first of all thanks a lot for the great library. Pardon me in advance if I might be overlooking at some details here but I need to your help to understand what is happening here and how to fix it.

Basically I have a stepper motor running flawlessly at various speeds and microsteps and I have a timer printed on LCD (through I2C) updated every second. This feat is to check how long the motor has been running and eventually use it also as a countdown and stop feat.

So far I have tested my code and I see that none of the actions I do (not many anyway) seem to cause the stepper to skip a beat, until I write something on the display. It doesn't matter if I am sending 1 single character or a full 16chars string.

I understand that those displays are damn slow but I thought that it would have been non-blocking so that I could send the string and use time for other stuff... I guess it is not the case, is it?

Is what I am trying to accomplish possible?

By the way I2CexpDiag reports:

LCD at address: 0x27 | config: P01245673H | R/W control: Yes

Speed is shown as:
i2c @100000:
ByteXfer 543uS
16x2 FPS 54.16
Ftime 18,46ms

i2c @400000
ByteXfer 198uS
16x2 FPS 148.46
Ftime 6.74ms

I know from some of your forum posts that I could use stronger pullups but I believe this is not of big impact performance wise.

Oh, and I am using Arduino Nano Every.

Thanks in advance

bperrybap, 2020-05-21:
Not sure what you mean by "blocking". 
 Do you mean blocking time from loop() or blocking interrupts causing servo issues?
I assume you are using the Servo library?
The Servo library uses interrupts, so even though Wire may be sending characters to the LCD, Servo interrupts can still occur. However......
Wire also uses interrupts, so it could be that the Wire library interrupts are occasionally delaying the Servo interrupts enough to cause some issues.

If you referring to blocking time from loop() until all the characters are sent to the LCD.

If you call the hd4480 library to print characters to the LCD using print() it uses the Print class print() function which loops calling the hd44780 library write() function, the Print class code will not return until all the characters have been handed to the library below it (one at a time), in this case hd44780.
That is how the Print class works. 

The hd44780_I2Cexp i/o class calls the Wire library to send the data to the LCD.
The Wire library is synchronous. While it does buffer up bytes for output, it does not run in the background  processing those bytes like Serial does. 
That is the reason print() will not return until all the characters have been sent to the display.

From LCDiSpeed you can see the time it takes per character so you can calculate about how long a print() will take based on how many characters you send.

bperrybap, 2020-05-21:
From a few years back:
https://forum.arduino.cc/index.php?topic=228364.0

federico77, 2020-05-21:
Despite the forum post talks about the servo being stationary instead of a stepper missing a beat (freezing so that it is noticeable) I can well see the issue similarities: it seems to lead to how the Wire library behaves which makes (sadly) a lot of sense.

Going back to a better description requested in your previous comment:

I am using the library [AccelStepper](http://www.airspayce.com/mikem/arduino/AccelStepper/) with its non blocking functions: I basically call stepper.runSpeed() during loop() and stepper.setSpeed() somewhere else in the code.

I started by using lcd.print() commands, then optimized along the way by only refreshing needed values when necessary. I also tried to remove the print() command and substitute it in most parts with straight lcd.write() with no benefits.

It is indeed clear that Print won't give back control until all chars are written but the sad part is that I don't see the glitch getting better or worse based on the number of characters: printing a simple "0" seems to do only slightly better (shorter ticks) than printing "hh:mm:ss" plus all the overhead added by conversions from millis, etc.

My idea at this point would be to execute the stepper & the rest of my code only when I2C is not engaged. To be honest I am not totally sure on how to do it and if it is done at a deeper level already now. Also it feels like a workaround...

What you suggest?

bperrybap, 2020-05-21:
Sorry. I misread the initial description of what s/w & h/w was being used.
I have no idea why I was thinking servo, even though clearly it says stepper.
That code stepper code does not use interrupts so it is synchronous so yes that code stepper is technically blocking as it does all the work when called as there nothing that is done in the background with ISRs.
It calls the functions to do the work in the foreground from being called from loop()
It relies/depends on the loop code calling it fast enough to keep up.
And so the user stepper functions that do the step work need to be quite short/fast.

That type of design is very fragile as depending on the needed timing, the rest of the entire system may need to not do very much and may have severe constraints on any other added capabilities such as controlling other devices such as an LCD or a sensor.
Also there can be lots of jitter in the timing of the calls to the stepper code since the foreground code  in loop() and other code called from loop() can take various amounts of time.
Depending on loop() being fast enough can work, but it is prone to potential issues since that type of design is so fragile.
If ISRs were used, then the code could interrupt and do the stepping when at the specific/desired and needed points in time, regardless of how the foreground works, what it is doing, or how long it may block.

What kind of timing is needed for the calls to run() and subsequent calls to the step functions?
The overall loop()  time between calls to run()  should be 1/2 or less of this time to ensure you don't miss a time slot.

So if using a 100Khz i2c bus clock and printing out HH:MM:SS on the display that will be 8 characters + a set character position or 9 LCD bytes transfered to the LCD that is roughly 5ms maybe a touch more for the formatting code so make it 6ms for safety
With that type of overhead, the run() should not expect to be called more than every 12 ms.
At 400Khz that reduces to 2ms LCD overhead and about 4-5ms between assured calls to run()

How often does run() need to be called?

federico77, 2020-05-23:
Hello, and thanks a lot for the quick, in-depth replies... that was actually faster than getting my next chance to get back to this project.

The AccelStepper library reports "Most API functions never delay() or block (unless otherwise stated)" and the stepper.runSpeed() function is not supposed to be blocking but I believe that does not mean that is using interrupts of any sort... it wouldn't require a .run() or .runSpeed() otherwise... am I right?

So you sparked an idea... I tried something different: https://github.com/laurb9/StepperDriver seems to have a non blocking mode and this indeed does works as stated but so far with a caveat:

if I start the rotation in the setup() and I keep it running with nothing in the loop except my LCD timer, everything runs smooth: sadly the library requires to set a number of steps/rotations in advance so I have to set it up with many rotations, check how many are still to be accomplished in the loop and bump up the counter.

Funny thing is... that every time I bump the rotations counter i have those brief hickups in the stepper.

Also to notice that the StepperDriver library has a function that allows to check how much time is left before the next action so it is easier to run the LCD refresh only if there is enough time for it.

At this point what I know is that AccelStepper is very cool but StepperDriver better fits my needs; your library is very nice despite I2C being (in general) quite crappy.

What I wonder is if there is a way to avoid that stepper hickup (I still am not sure of where it is coming from) but I think this would not matter for this thread.

bperrybap, 2020-05-23:
> What I wonder is if there is a way to avoid that stepper hickup

Yes, but it may not be possible with s/w that uses that type of synchronous "polling" design.
The "hickup" is likely because the control software running the motor can not ALWAYS get in and do its thing when it needs to. Also, that code has some comments in it that it doesn't really track time properly and can have issues (like losing track of where the motor is) if things don't happen on schedule as it doesn't (and really can't) account for lost time and or motor momentum from not running at the appropriate time.

Whenever there is a need for things to happen within a very time critical window of time, particularly when using a generic system that may be doing other things, then there must be some way to ensure that the time critical pieces of code can run when needed. A common of way of doing that it is to use interrupts so that what ever is running can be pre-empted to do the time critical operations.

Using techniques like polling or periodic calling of control functions like this library uses is unreliable since there is no way to always ensure that calls to it ALWAYS occur fast enough.
Depending on what is is going on in the system, the time between calls/polls can be too long which creates issues which is why I said that type of design is so fragile.

bperrybap, 2020-05-23:
Just for clarification as it may not have been clear, when I said, 

> Using techniques like polling or periodic calling of control functions like this library uses 

I meant the issue is  in the AccelStepper  library not the hd44780 or Wire libraries.

federico77, 2020-05-23:
It was lound and clear from the beginning, thank you! :)

closed 2023-04-19 by bperrybap