duinoWitchery/hd44780 #17: Missing "Wire.h" include

bohdanyurov-gl, 2020-08-10:
Looks like hd44780_I2Cexp.h is missing Wire.h include.

bperrybap, 2020-08-10:
It is not missing.
It is intentionally not included because a user may need or want to include a different named header file for "Wire" support.
For example if a user wants to use a software wire implementation, they could do that by using:
```
#include <SoftwareWire.h>
```
in their sketch rather than <Wire.h>
You can see this in the hd44780_I2C i/o class example SoftwareWire
If hd44780_I2Cexp.h was including the Wire.h header file it would preclude use cases like this.

bohdanyurov-gl, 2020-08-10:
Makes sense. That is fine for Arduino sketches, but not for PlatformIO, though.
Thanks for the answer, anyway.

bohdanyurov-gl, 2020-08-10:
Nevermind, it still works when specifying dependencies correctly. Stupid me.

closed 2020-08-10 by bohdanyurov-gl