duinoWitchery/hd44780 #21: Can't compile for Digispark

Barabba11, 2021-02-22:
Hi! Im usng the following example found here:
https://forum.arduino.cc/index.php?topic=579136.0
```

#include <Wire.h>
#include <hd44780.h>                       // main hd44780 header
#include <hd44780ioClass/hd44780_I2Cexp.h> // i2c expander i/o class header

hd44780_I2Cexp lcd; // declare lcd object: auto locate & auto config expander chip

// LCD geometry
const int LCD_COLS = 20;  // replace with your display parameters
const int LCD_ROWS = 4;

void setup()
{
   lcd.begin(LCD_COLS, LCD_ROWS);
   lcd.print("Hello, World!");
}

void loop()
{
   static unsigned long timer = 0;
   unsigned long interval = 500;
   if (millis() - timer >= interval)
   {
      timer = millis();
      lcd.setCursor(0, 1);
      lcd.print("millis =        ");
      lcd.setCursor(8, 1);
      lcd.print(millis());
   }
}
```

But I have this log:
`
C:\Users\Admin\Desktop\arduino-1.8.5\arduino-builder -dump-prefs -logger=machine -hardware C:\Users\Admin\Desktop\arduino-1.8.5\hardware -hardware C:\Users\Admin\AppData\Local\Arduino15\packages -tools C:\Users\Admin\Desktop\arduino-1.8.5\tools-builder -tools C:\Users\Admin\Desktop\arduino-1.8.5\hardware\tools\avr -tools C:\Users\Admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Admin\Desktop\arduino-1.8.5\libraries -libraries C:\Users\Admin\Desktop\oxi\Arduino\libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10813 -build-path c:\temp\arduino_build_794276 -warnings=none -build-cache c:\temp\arduino_cache_120680 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=C:\Users\Admin\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=C:\Users\Admin\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.avr-gcc.path=C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -verbose C:\Users\Admin\Desktop\Oxi\Arduino\Digispark-LCD-I2C\Digispark-LCD-I2C.ino
C:\Users\Admin\Desktop\arduino-1.8.5\arduino-builder -compile -logger=machine -hardware C:\Users\Admin\Desktop\arduino-1.8.5\hardware -hardware C:\Users\Admin\AppData\Local\Arduino15\packages -tools C:\Users\Admin\Desktop\arduino-1.8.5\tools-builder -tools C:\Users\Admin\Desktop\arduino-1.8.5\hardware\tools\avr -tools C:\Users\Admin\AppData\Local\Arduino15\packages -built-in-libraries C:\Users\Admin\Desktop\arduino-1.8.5\libraries -libraries C:\Users\Admin\Desktop\oxi\Arduino\libraries -fqbn=digistump:avr:digispark-tiny -ide-version=10813 -build-path c:\temp\arduino_build_794276 -warnings=none -build-cache c:\temp\arduino_cache_120680 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.micronucleus.path=C:\Users\Admin\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.micronucleus-2.0a4.path=C:\Users\Admin\AppData\Local\Arduino15\packages\digistump\tools\micronucleus\2.0a4 -prefs=runtime.tools.avr-gcc.path=C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -prefs=runtime.tools.avr-gcc-4.8.1-arduino5.path=C:\Users\Admin\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\4.8.1-arduino5 -verbose C:\Users\Admin\Desktop\Oxi\Arduino\Digispark-LCD-I2C\Digispark-LCD-I2C.ino
Using board 'digispark-tiny' from platform in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7
Using core 'tiny' from platform in folder: C:\Users\Admin\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7
Detecting libraries used...
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "c:\\temp\\arduino_build_794276\\sketch\\Digispark-LCD-I2C.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for Wire.h: [Wire Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire Wire@1.0]
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\Wire" "c:\\temp\\arduino_build_794276\\sketch\\Digispark-LCD-I2C.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for hd44780.h: [hd44780@1.3.2]
ResolveLibrary(hd44780.h)
  -> candidates: [hd44780@1.3.2]
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\Wire" "-IC:\\Users\\Admin\\Desktop\\oxi\\Arduino\\libraries\\hd44780" "c:\\temp\\arduino_build_794276\\sketch\\Digispark-LCD-I2C.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\Wire" "-IC:\\Users\\Admin\\Desktop\\oxi\\Arduino\\libraries\\hd44780" "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\Wire\\USI_TWI_Master_Wire.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\Wire" "-IC:\\Users\\Admin\\Desktop\\oxi\\Arduino\\libraries\\hd44780" "C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\Wire\\Wire.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
"C:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\4.8.1-arduino5/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -w -x c++ -E -CC -mmcu=attiny85 -DF_CPU=16500000L -DARDUINO=10813 -DARDUINO_AVR_DIGISPARK -DARDUINO_ARCH_AVR "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\cores\\tiny" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\variants\\digispark" "-IC:\\Users\\Admin\\AppData\\Local\\Arduino15\\packages\\digistump\\hardware\\avr\\1.6.7\\libraries\\Wire" "-IC:\\Users\\Admin\\Desktop\\oxi\\Arduino\\libraries\\hd44780" "C:\\Users\\Admin\\Desktop\\oxi\\Arduino\\libraries\\hd44780\\hd44780.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
[... cut ...]
Usata: C:\Users\Admin\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\Wire
Non usata: C:\Users\Admin\Desktop\oxi\Arduino\libraries\Wire
Uso la libreria Wire nella cartella: C:\Users\Admin\AppData\Local\Arduino15\packages\digistump\hardware\avr\1.6.7\libraries\Wire (legacy)
Uso la libreria hd44780 alla versione 1.3.2 nella cartella: C:\Users\Admin\Desktop\oxi\Arduino\libraries\hd44780 
exit status 1
Errore durante la compilazione per la scheda Digispark (Default - 16.5mhz).
`
How to solve it? Thanks

bperrybap, 2021-02-22:
I need to know which h/w platform you have added/installed and which board type are you using so I can replicate the exact environment you are using.

I do see an issue/difference with the digistump platform from here: http://digistump.com/package_digistump_index.json
But that may not be what you are using.

The core used by the Digispark board in the digitistump core defines the F() macro and associated data type differently than every other core so it creates an issue for the hd44780 library.
This is technically a core compatibility issue, not really a hd44780 library issue.
I can potentially work around it but I need to test with the exact environment you are using.

bperrybap, 2021-02-22:
For now you can comment line 152 in hd44780.h
`//	size_t __attribute__ ((error("println() is not supported"))) println(const __FlashStringHelper *);`

Barabba11, 2021-02-24:
thank you mate, you really friendly!
Well I?ve solved isntalling anoither board, Attinycore, but to be honest I haven't compiled your library with that. By chances I had another sketch with LiquidCrystal_I2C (that didn't work too) and it run from first time.. so I didn't want to change as I've to finish the project. Apologize me. You can close the issue if you think, thank you again

bperrybap, 2021-02-24:
It is still an issue I need to resolve before closing this issue.

Can you tell me which platform and which board type you were using so I can make sure that any workaround I add in works for that environment.
I also need to know this information so I can update the hd44780 library documentation and list of known issues.

bperrybap, 2021-02-24:
I think you were using the board package I mention but 
I need to know for sure the url of the bored package you added to
the Arduino preferences "Additional Board Manager URLs".
and then then board type you selected.
Can you please provide me that information?

bperrybap, 2021-09-02:
Can you provide the information about the specific board and board package you were using?

Barabba11, 2021-09-02:
Omg, so much time passed and so many things I did. My friend sorry, that I can't help you. For sure I used digispark, so attiny85 16mhz. I know I got it working, probably I used another library.