duinoWitchery/hd44780 #22: I want to use it on a raspberry pi

9acca9, 2021-03-02:
Hi.
Sorry for my ignorance, but... if i want to use this module?? What i need to do to try it on a raspberry pi??
i download what it is here and compile it??
Can you tell me how? i see that you talk about Arduino.
Thanks for your help!

bperrybap, 2021-03-02:
Sorry, I guess I should update the main readme to  indicate that code is used to create an Arduino library.
It is designed to use the Arduino environment.
The Arduino environment is very primitive and normally runs directly on a micro controller.
The Raspberry Pi normally runs linux. linux is a very feature rich environment. It is very different than the primitive embedded environment provided by Arduino.
The hd44780 Arduino library code could be updated to work on linux on the Pi using the GPIO port but as is the code will not work as is, in on linux.

bperrybap, 2021-03-02:
Actually, it already says it is an Arduino library right in the first sentence of the Library Overview:
> hd44780 is an extensible Arduino LCD library for hd44780 based LCD displays.

bperrybap, 2021-09-02:
Closing since library is for Arduino and is currently not suitable for Raspberry Pi.

closed 2021-09-02 by bperrybap