duinoWitchery/hd44780 #23: Requesting more stable initialisation routine for OLED

jazzyjoepass, 2021-04-10:
First, amazing job well done.

I'm using a Raystar 40x2 OLED display with the common PCF8574 I2C backpack adapter. This HD44780 library works so much better than the common LiquidCrystal_I2C library. I'm not sure what has been employed to improve interfacing with OLED displays.

I have read somewhere that OLEDs need special attention during initialisation or reset. See post #12 and #9 on https://forum.arduino.cc/index.php?topic=91659.0. 

I notice that on rare instances, when I upload my Sketch, the OLED will display garbled images. If I was to unplug and replug the Arduino USB cable causing a power reset, the OLED displays correctly. Sometimes Row 1 and 2 is reversed - what was supposed to be displayed on Row 1 ends up on Row 2 and vice versa. 

The LiquidCrystal_I2C library fails miserably interfacing with OLEDs and garbled images are common and happens 90% of the time. I'm looking forward to a robust/complete solution. Thanks for the work. 

:)

bperrybap, 2021-04-10:
I'll respond back in the linked Arduino Forum thread.

bperrybap, 2021-04-10:
Actually, I just noticed that the thread you pointed to is VERY old. 

In going through the thread, and looking at the datasheets, I see some issues.
The devices have differences between them and there are issues in the command sets of the devices.
Things like the clear command doesn't work the same way as it does on a real hd44780 display. 
And the timing in some of the datasheets is obviously wrong.
Also attempting to use font bits in the FunctionSet instruction can 

Overall, after looking at the datasheets, and the videos, it looks to me like case of nibble synchronization being lost due to violating the device timing.
It could be related to the timing of the clear instruction or other instructions as depending on which datasheet is looked at as the instruction timing is considerably longer than the typical hd44780 display.
Adafruit created a special version of the LiquidCrystal library that attempts to check for busy status, rather than using fixed timing.
But they screwed it up and it has issues. I would not use that library as the way they wrote it creates a short on the data pins while polling busy. This could damage the Arduino or the OLED.

The hd44780 library can easily be configured for different timing.

Please start a new thread for this and post a link to it in this issue and I will respond further in that thread.

jazzyjoepass, 2021-04-11:
You mean to start another thread in your Github repository or Arduino forum?

Yes a lot of people have mentioned that timing on OLED is longer.
I have minimal experience with Arduino and Raspberry - just a hobby.

I'm in a group of repair techs who occasionally have to deal with dead displays and have tried using OLEDs as replacements.
Some have used OLEDs successfully, some have not, depending on the equipment in question.
But we all know OLEDs behave slightly different from LCD and it's an issue of timing. And usually happens when the screen is cleared or initialisation stage. I'm trying to understand more about OLEDs hence my experiments with Arduino.

Thanks for your replies.

Kind regards,

Mike Foo.

bperrybap, 2021-09-02:
Is this still an issue?