duinoWitchery/hd44780 #24: redefinition of 'hd44780_I2Clcd lcd'

chakahamilton, 2021-04-18:
Hi i am getting this error when i run the linewrap example
redefinition of 'hd44780_I2Clcd lcd' i uncommented this line so i could add my own address
hd44780_I2Clcd lcd(i2c_addr); // use device at this address

bperrybap, 2021-04-18:
You have not provided enough information me to be able to fully understand what you have done or to be able to reproduce what you have done.
Accurate details are very important. For example, there is no linewrap example, but there is a LineWrap example.
If you are using the hd44780_I2Clcd i/o class LineWrap example and simply uncommitted the line you mention, then, yes, the error you mentioned is to be expected since you can only define the lcd object once.

If you uncomment one object definition, you must make sure that all others are commented out.

chakahamilton, 2021-04-19:
I do apologize for the generic lack of information. I was able to figure it out shortly after posting but forgot to close this issue.

Best regards,

closed 2021-04-19 by chakahamilton

bperrybap, 2021-04-19:
So what was the issue?

chakahamilton, 2021-04-19:
It was a simple matter of me commenting out a line towards the top of the code.