duinoWitchery/hd44780 #25: Scrolling Veristically from bottom to Top

chakahamilton, 2021-04-22:
Hi,
I am using the following hardware: 
1) **16x4 LCD with the PCF8574T I2C backpack** 
2) ESP32

I am able to get the library to work including wrap and scroll left to right, Is it possible using this library to  scroll vertically from the bottom line to the top?

Regards

bperrybap, 2021-04-22:
I do not understand what you are asking.

chakahamilton, 2021-04-22:
**Clarification:**
So normally when using scrollDisplayLeft() or scrollDisplayRight() the text will either go from left to right or right to left. what I want to do is the following:

I set up a system to receive and sms if the message is too long i.e it fills up the screen, Id like it to scroll the old text up through each line (bottom of lcd to top) to allow for additional parts of the message to continue coming it. (You may or may not remember the old-school alpha numeric pagers.) like that.

I hope that makes sense.

Regards.

bperrybap, 2021-04-22:
The old-school pager I had back in the late 80s to early 90s only had a single line that would scroll the messages right to left.

IMO, the scrollDisplayLeft() and scrollDisplayRight() which use capabilities built in to the display can really only be used on single line displays.

What you are likely wanting is what I call "terminal" mode.
This is when long lines are wrapped to the line below and the text scrolls up all lines when the bottom line wraps to make room for the new line just like a text terminal.
This is on my list to do, but I have not added it yet.
I've been tossing around a few different ways to implement it.
They all have advantages / disadvantages.

One way uses a virtual display buffer so that it can work LCDs and interfaces that do not support reading.
Another is to only support it on displays that support reading as it is relatively simple and does not require any additional ram storage.

The next thing to decide is whether to handle it in the existing i/o class or to create a wrapper template class.
The advantage of doing it in a wrapper class, is that if the capability wasn't needed then there is no code or ram penalty.

In the mean time, for an applications that needs to print long text messages, you can do something like what is done in the Serial2LCD example.
It clears the display for each message.
It has been updated to do printing of a the message on the display while the message is still coming in to allow for longer messages than the serial buffer can hold.
Its is currently in the alpha branch but not yet in the master branch as it has not yet been released.
https://github.com/duinoWitchery/hd44780/blob/alphadevel/examples/ioClass/hd44780_I2Cexp/Serial2LCD/Serial2LCD.ino

bperrybap, 2021-04-22:
I did a terminal mode in my openGLCD library.
And in the openGLCD library, you can declare multiple virtual displays on the physical screen.
This allowed some interesting capabilities like being able to create tiny displays of say a fixed width of characters that could be used to update a field on the physical display by simply writing to that virtual display. 

Doing a terminal mode is more complicated than it sounds at first. There are lots of little things that have to be addressed like end of line handling. This is ridiculously complicated because of Microsoft and Apple not using the unix line terminator that was defined a few decades before they created their own way of handling it.
It is easy to support any 2 of them but having to support all 3 automatically is not possible so it has to be configured.
Then there are some things that need to be done to make the printing on the very bottom line work the way people expect vs the 
way it would work if just blindly handling the line endings as sent.
And many of not most real text terminals screwed up their handling of line wrapping right at the very last character on the line, and in particular on the last character of the last line on the display.

But I did lots of serial kernel drivers for unix back in the early 80s and then did lots of terminal emulation s/w so I'm very familiar with all the little pitfalls that many implementations often don't do correctly.

chakahamilton, 2021-04-22:
Wow thanks for the in-depth explanation. Yes terminal mode. I'm glad it has a real term than scroll vertical.  Ill give example code you provided go. I am glad I found your Git, it is the most feature rich library I have found to date. I appreciate the work you've done on it. 

Thank you again for you time.
ps [this was my first 📟 ](https://images-na.ssl-images-amazon.com/images/I/41%2B1f-m41LL._AC_.jpg) where I got the idea from.

Regards.

closed 2021-04-22 by chakahamilton

bperrybap, 2021-04-23:
That is "advanced" compared to the pager I had.
The one I had used a 12 character display of digits only and had a single button.