duinoWitchery/hd44780 #26: LCD "Busy"

OutlawzHD, 2021-04-28:
What does this mean? While running the IC2expDiag to figure out what is going on with my LCD screen since I have tried dozens of libraries and can't seem to print anything. I discovered this library and it returned that the lcd was "busy" and can not be used. Any suggestions?

bperrybap, 2021-04-28:
github issues are for reporting issues, "bugs", not for support.
The Arduino forum is for support and questions.

Unfortunately, at this point in time, getting support is a bit difficult since the arduino.cc boys got a brilliant idea to completely redo the forum site and make it less of a technical board and more of a social discussion board that offers "likes" and "badges" for participation. Along with that came many changes that broke things and made the site much more difficult to use for technical users that provided support to the community.
So I'm currently not visiting it anymore since it is so @#$@#%! to use now.

I'll briefly address this question here.
If BUSY is stuck it means what it says.
The diag tool has looked at the BUSY status bit and it is stuck high indicating that the LCD is busy and cannot accept further instructions.
This typically this is not the case but indicates an issue somewhere else.
This can be caused by several things. From bad soldering of the backpack to the LCD, voltage issues, bad soldering on the PCF8574 chip pins on the backpack module or a bad PCF8574 chip.
Is is also possible that due to some kind of issue, the library mis detected the pin wiring used by the backpack so the library is looking at the wrong signal pin on the LCD.
Without seeing the diagnostic output or the soldering it is impossible to say.
But the most likely cause of an issue like this is a soldering issue, of either the pins to the LCD or the PCF8574 pins on the module.

OutlawzHD, 2021-04-28:
> github issues are for reporting issues, "bugs", not for support.
> The Arduino forum is for support and questions.
>
> Unfortunately, at this point in time, getting support is a bit difficult since the [...]

Well the backpack isn’t soldered to the LCD so I’m assuming that is the issue. Thanks I’ll either look into learning soldering or just get another LCD.

bperrybap, 2021-04-28:
You must solder the backpack to the LCD. If not, there will be issues.
For a short/brief test, you can sometimes get away with holding the backpack at an angle to put pressure on the pins.
I"ve done that for some brief testing, but you definitely can't count on it working for anything but a brief test.

Soldering is a good skill to learn. For this type of small pin soldering, the key is to get a soldering iron with a pointy small tip and try to get the old resin core lead based solder like 60/40 in a small diameter like 1 to 2 mm. That old led based stuff works best
Also, using a smaller diameter solder is easier since it is much  easier to control and helps avoid getting too much solder on the connection.
Then practice on some things that are not important till you get the hang of it.

I would highly recommend getting a soldering station and a helping hands.
Can you solder without them, yes, but they make soldering so much easier, especially having a place to hold the iron.
I use a 50w solder station with a variable heat control.
They are not that expensive, starting at about $25 USD.
I have this one: https://www.amazon.com/Velleman-VTSS5U-Low-Cost-Soldering-150-480%C2%B0C/dp/B000I40HFQ/ref=sr_1_5?dchild=1&keywords=velleman+soldering+station&qid=1619629018&sr=8-5
I solder with it typically at about 75% for this type of soldering.

You can get them with more features like tip temperature, and a place for the solder.
But a basic one will work fin.

Also, I recommend getting "help hands".
https://www.amazon.com/Elenco-Helping-Magnifying-Case-Iron-Chrome/dp/B0002LLWYK/ref=sr_1_6?crid=151WWEK02LJU4&dchild=1&keywords=helping+hands+soldering&qid=1619629205&sprefix=helping+hands+sol%2Caps%2C181&sr=8-6
They come in many different styles and designs.

If your eyes are not so young anymore, then I'd recommend getting one with a magnifying glass (biggest you and find/afford), that has a light/led on it. It can make a big difference - it does for me since my eyes are not as good as they were 50 years ago when I first starting soldering.

bperrybap, 2021-09-02:
Not really a library s/w issue.

closed 2021-09-02 by bperrybap