duinoWitchery/hd44780 #29: Using hd4470 library with the Arduino DUE's second I2C port (SDA1/SCL1)

mowmow65, 2021-08-20:
Is there a config option that allows me to use this library with the SDA1/SCL1 pair on the Arduino DUE? This is the second I2C port that connects to pins 9 & 70 of the SAM3X8E processor and the extreme end two pins next to AREF on the DUE's header.

I'd like to use both I2C ports, but only the second port (SD1A/SCL1) would be connected to an LCD. I'm using a PCF8574 based backpack on the LCD.

FWIW, I'm currently setting the primary port speed to 400K using Wire.setClock(400000) to speed things up - maybe Wire1.setClock(400000) will be needed for the second port.

bperrybap, 2021-08-20:
For the time being, you should be able to do a hack to trick the library into using another i2c bus by using a macro.

So to use Wire1 instead of Wire you would this:
```
#include <Wire.h>
#include <hd44780.h>                       // main hd44780 header
#define Wire Wire1  // hack to use Wire1 object instead of Wire object
#include <hd44780ioClass/hd44780_I2Cexp.h> // i2c expander i/o class header
#undef Wire            // disable macro for any other code
```

The #define will trick the hd44780_I2Cexp i/o class to reference the Wire1 object instead of the Wire object
and the undef will turn that off so you can use the Wire object again for any other library or sketch code.

The final solution that I'll be adding will be to make the class templated so that the user can specify the Wire object in the constructor declaration. For now, this "hack" should work.

Note: I can't test this as I don't have any DUE hardware.
But I have tested this same mechanism for some other software i2c libraries that use a different object name than Wire and it works fine.

Also note, You will not be able to use/run any of the examples provided unless you modify them.
If you are wanting to run I2Cexpdiag sketch example on Wire1 then you do the same modification in the sketch but don't put in the undef line as the i2CexpDiag code also needs to talk to the wire object being used.

mowmow65, 2021-08-21:
Brilliant - that did the trick. Wasn't sure if you were bypassing Wire in any way.
40 years of pecking at the keyboard and this was my first deliberate use of #undef.
Thanks so much for the sonic response. Feel free to close this thread if that's what I should have done at this point.

bperrybap, 2021-08-21:
I'll close this but  I'll create a new issue for adding support for any/all Wire library implementations. It is already on my list of things to do/fix but there currently isn't an issue for it.

closed 2021-08-21 by bperrybap