duinoWitchery/hd44780 #32: choose different SDA and SCL pins

Irgendwer008, 2022-03-01:
is it possible to change the SDA and SCL pins for i2c? i am using an STM32F103C8T6 Board and with the normal pins everything works fine but i'd like to change the pins from the standart Wire-library pins (I think PB7 and PB6) to PB11 and PB10. Is that possible somehow?

bperrybap, 2022-03-01:
This is outside the hd44780 library as the hd44780 library does not have any code to control the i2c interface or its pins.

It appears that the STM32 platform has Wire library API extensions for setting the pins.
Wire.setSDA(sdapin);
Wire.setSCL(sclpin);

Or both at same time:
Wire.begin(sdapin, sclpin);

https://www.stm32duino.com/viewtopic.php?t=1047

You will need to configure the pins before you call the hd44780 lcd.begin() function.

closed 2022-03-01 by bperrybap

Irgendwer008, 2022-03-01:
alright thanks, i'll look into it (but tomorrow its 2am here lol)
but thanks for the fast reply!