duinoWitchery/hd44780 #33: plain simple bare minimum democode

StefanL38, 2022-07-17:
Hi everybody

I came across multiple postings that praise the hd4478-library as the "**best**" LCD-library to use
because it is able to find the configuration itself.

Well this might be true. When I was looking through the example-codes I just found high complex specialised sketches
that all seem not to demonstrate a bare minimum democode **well suited for beginners** that use an Ardunio-Uno or Arduino Mega 2560

If using the HD4478-library requires so much configuration as shown in the example-files then it is not beginner-friendly.
Then any advantage of autoconfiguration is vaporised by the too complexe example-codes.

It will take less time to do a quick I2C-Scanner.ino-run and to adjust the I2C-adress and then use the liquied_chrystal_I2C-library

I guess for an expert that knows the hd44780-library very well it will just take 10 minutes or even less to write down a bare-minimum demo-code that runs on arduino Unos and Arduino-Mega 2560

If the library is really wel suited to work with a lot of different LCDs and a lot of different platforms this should be even true for 
SAMD-based microcontrollers, ESP8266 and ESP32 etc. etc.

But maybe I'm wrong with assuming this. 
Can some expert or the author shine a light on this?

best regards Stefan

StefanL38, 2022-07-17:
I was asking for such a code in the Arduino-forum and user groundfungus posted a demo-code 
`
// hd44780 library see https://github.com/duinoWitchery/hd44780
// thehd44780 library is available through the IDE library manager
#include <Wire.h>
#include <hd44780.h>                       // main hd44780 header
#include <hd44780ioClass/hd44780_I2Cexp.h> // i2c expander i/o class header

// if there should be an additional I2C-devices on the I2C-Bus 
// which has a lower I2C-adress it might happen that this other devices I2C-adress 
// will be chosen as the I2C-adress of the LCD
// if this happends add the I2C-adress to the constructor of the HD44780
// example instead of writing hd44780_I2Cexp lcd;
// add the I2C-adress hd44780_I2Cexp (0x27);
hd44780_I2Cexp lcd; // declare lcd object: auto locate & auto config expander chip

// LCD geometry
const int LCD_COLS = 16;
const int LCD_ROWS = 2;

// easy to use helper-function for non-blocking timing
boolean TimePeriodIsOver (unsigned long & startOfPeriod, unsigned long TimePeriod) {
  unsigned long currentMillis  = millis();
  if ( currentMillis - startOfPeriod >= TimePeriod ) {
    // more time than TimePeriod has elapsed since last time if-condition was true
    startOfPeriod = currentMillis; // a new period starts right here so set new starttime
    return true;
  }
  else return false;            // actual TimePeriod is NOT yet over
}

void setup() {
   lcd.begin(LCD_COLS, LCD_ROWS);
   lcd.clear();
   lcd.print("Hello World");
   lcd.setCursor(0, 1);
   lcd.print("Millis ");
}

void loop() {
   updateLCD();
}

void updateLCD() {
   static unsigned long lcdTimer = 0;

   if ( TimePeriodIsOver(lcdTimer,500) ) { // check if more than 500 milliseconds have passed by
      // if 500 milliseconds HAVE passed by
      lcd.setCursor(8, 1);
      lcd.print("       "); // overwrite old data
      lcd.setCursor(8, 1);  // reset the cursor
      lcd.print(millis());
   }
}
`

So what do you think about adding such demo-codes?
best regards Stefan

bperrybap, 2022-07-17:
There is no need to add such example(s) to the hd4780 library as examples for all the i/o classes including for the hd44780_I2Cexp i/o class are already there.
The hd44780 library already includes a "HelloWorld" type simple/minimal example similar to the one you posted for each i/o class as well as many other examples to demonstrate the use of the library for the given i/o class.

Unfortunately, you decided to not read any of the documentation before trying to use the library so you are not looking in the proper place for the examples for the hd44780_I2Cexp i/o class.
i.e. the examples are partitioned by i/o class, and the examples for each i/o class are grouped together under a directory by the i/o class name. 
note: I will say you are not alone in making this mistake - so I will be making some changes to the library structure to try to minimize the confusion for those users that choose to not read any of the documentation before diving right in.

The hd44780 library has quite a bit of documentation. Much more than most libraries.
There is a **Documentation** sketch included in the library that has an overview as well as links to additional information and documentation, including a link to API documentation and a link to the hd44780 wiki.
There is also additional documentation on the github site in the wiki:
https://github.com/duinoWitchery/hd44780/wiki

Also each example sketch for all the i/o classes are highly commented to provide the user with additional information about the sketch and how to use various library capabilities.
Some examples demonstrate specific library capabilities or specific i/o class capabilities so different i/o classes may have different examples.

I would recommend spending a few minutes looking at the hd44780 library documentation to get an understanding of how library is partitioned and where the examples for each i/o class live.

As noted in the wiki page for the hd44780_I2C exp i/o class,
https://github.com/duinoWitchery/hd44780/wiki/ioClass:-hd44780_I2Cexp
the first sketch that should always be run when first installing and testing a device that uses the hd44780_I2C exp i/o class, is the **I2CexpDiag** sketch.
This will test that everything is installed and working correctly and run h/w tests on the i2c signals and the LCD internal memory.

-----------------

Given that quite a few people choose to not read the documentation and get confused by the special purpose code sketches (which are not i/o class example sketches).
I will in the near future push out a 2.0 release that moves some things around and renames a few of the example directories to try to make the location of the i/o class examples at bit more obvious as well as hiding the special purpose purpose "sketches" in the hd44780examples directory.
The hd44780example special purpose "sketches"  will longer look like sketches and will no longer be visible in the IDE.

This has been on my todo list for quite some time.

closed 2022-07-17 by bperrybap