duinoWitchery/hd44780 #39: status -4 for even a most simple sketch?

StefanL38, 2023-04-19:
This sketch devivers status -4
```
#include <Wire.h>
#include <hd44780.h>
#include <hd44780ioClass/hd44780_HC1627_I2C.h>

hd44780_HC1627_I2C lcd;

int selected = 0;

const int LCD_COLS = 20;
const int LCD_ROWS = 4;

///
/// START PROCEDURE
///
void setup() {
	Serial.begin(9600);
    Serial.println("setup STARTED!");
  
    int status;
    status = lcd.begin(LCD_COLS, LCD_ROWS);
    Serial.println("lcd.begin done");
    if (status) { 
        Serial.println("entering if (status)");
        Serial.print("value of status=");
        Serial.println(status);    
        hd44780::fatalError(status);
        Serial.println("LCD Initialized!");        
    }
  
    lcd.clear();
    Serial.println("lcd.clear done");
    lcd.setCursor(0, 0);
    Serial.println("lcd.setCursor done");
    lcd.print("Hello!");
    Serial.println("lcd.print done");
}

///
/// UPDATE PROCEDURE
///
void loop() {

}
```

while this code with the standard liquiqchristal I2C -lib works
```
#include <Wire.h>
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27,40,4);  

int selected = 0;

const int LCD_COLS = 20;
const int LCD_ROWS = 4;

///
/// START PROCEDURE
///
void setup() {
	Serial.begin(9600);
    Serial.println("setup STARTED!");
    lcd.init();                      // initialize the lcd

    // Print our characters on the LCD
    lcd.backlight();  //Backlight ON if under program control
    Serial.println("lcd.backlight done");
   
    lcd.clear();
    Serial.println("lcd.clear done");
    lcd.setCursor(0, 0);
    Serial.println("lcd.setCursor done");
    lcd.print("Hello!");
    Serial.println("lcd.print done");
}

///
/// UPDATE PROCEDURE
///
void loop() {

}
```

bperrybap, 2023-04-19:
If one of the many LiquidCrystal_I2C libraries you found is working, then the LCD device you have has a backpack using an PCF8574 i/o expander chip
The hd44780 library hd44780_HC1627_I2C i/o class you used in the first example is for LCD devices that use the HC1627 chips.

The h/w and communication interfaces for those two devices are very different.

If you have an LCD with a PCF8574 i/o expander chip on a backpack then attempting to use the  hd44780_HC1627_I2C i/o class will fail since the library will be trying to communicate with a HC1627 device.

For a LCD with a PCF8574 i/o expander based backpack, the proper hd44780 library i/o class is hd44780_I2Cexp
This is indicated in the hd44780 library documentation.
You can get to the documentation & Wiki directly from the IDE GUI via the included **Documentation** sketch.
Here is direct link to the **H/W support** section of hd44780 library main Readme:
https://github.com/duinoWitchery/hd44780#hw-support
Here is link to the WikI which has additional information and sub links
https://github.com/duinoWitchery/hd44780/wiki
See the section: **hd44780 i/o classes**
which has links to the wiki pages for various i/o classes.
You will want to read the documentation for the hd44780_I2Cexp i/o class:
https://github.com/duinoWitchery/hd44780/wiki/ioClass:-hd44780_I2Cexp

bperrybap, 2023-04-19:
I'll close this out as it is caused by user error vs an actual issue in the library.

closed 2023-04-19 by bperrybap