duinoWitchery/hd44780 #40: Arduino Nano 168: Sketch uses 106% of program storage space

boomsya, 2023-07-20:
_#include <Wire.h>
#include <hd44780.h>                       // main hd44780 header
#include <hd44780ioClass/hd44780_I2Cexp.h> // i2c expander i/o class header_
**RESULT:**
Sketch uses 15272 bytes (106%) of program storage space. Maximum is 14336 bytes.
Global variables use 587 bytes (57%) of dynamic memory, leaving 437 bytes for local variables. Maximum is 1024 bytes.

_#include <Wire.h> 
#include <LiquidCrystal_I2C.h>:_
**RESULT:**
Sketch uses 12434 bytes (86%) of program storage space. Maximum is 14336 bytes.
Global variables use 529 bytes (51%) of dynamic memory, leaving 495 bytes for local variables. Maximum is 1024 bytes.

bperrybap, 2023-07-20:
Not sure what you have done since you have not described what you have done or included the source code to the Arduino sketch that you are compiling.

If I build the hd44780_I2Cexp example sketch HelloWorld for the Arduino Nano 168
It is around 8k on both IDE 1.8.19 and IDE 2.1.1

> Sketch uses 8042 bytes (56%) of program storage space. Maximum is 14336 bytes.
> Global variables use 538 bytes (52%) of dynamic memory, leaving 486 bytes for local variables. Maximum is 1024 bytes.

Please provide exact details to be able to replicate your results.

boomsya, 2023-07-21:
I have deleted source code based on hd44780 because diffrent +20% of Nano storage is very big different
In the end I have optimized source code of program to minimize redraw count of LCD parts

closed 2023-07-21 by boomsya